orangepi-xunlong/wiringOP-Python #2: impossible to install

techaura, 2020-08-06:
WiringPi/wiringPi/wiringPi.c: In function ‘digitalWrite’:
WiringPi/wiringPi/wiringPi.c:1631:5: error: ‘version’ undeclared (first use in this function); did you mean ‘termio’?
  if(version == ORANGEPI) {
     ^~~~~~~
     termio
WiringPi/wiringPi/wiringPi.c:1631:5: note: each undeclared identifier is reported only once for each function it appears in
WiringPi/wiringPi/wiringPi.c:1631:16: error: ‘ORANGEPI’ undeclared (first use in this function); did you mean ‘ERANGE’?
  if(version == ORANGEPI) {
                ^~~~~~~~
                ERANGE
WiringPi/wiringPi/wiringPi.c:1663:4: warning: implicit declaration of function ‘OrangePi_digitalWrite’; did you mean ‘digitalWrite’? [-Wimplicit-function-declaration]
    OrangePi_digitalWrite(pin, value);
    ^~~~~~~~~~~~~~~~~~~~~
    digitalWrite
WiringPi/wiringPi/wiringPi.c: In function ‘pwmWrite’:
WiringPi/wiringPi/wiringPi.c:1706:9: error: ‘pwmmode’ undeclared (first use in this function); did you mean ‘pinMode’?
     if (pwmmode == 1)//sycle
         ^~~~~~~
         pinMode
WiringPi/wiringPi/wiringPi.c:1712:15: error: ‘MAX_PIN_NUM’ undeclared (first use in this function)
     if (pin < MAX_PIN_NUM) // On-Board Pin needto fix me Jim
               ^~~~~~~~~~~
WiringPi/wiringPi/wiringPi.c:1729:17: warning: implicit declaration of function ‘sunxi_pwm_get_period’ [-Wimplicit-function-declaration]
         a_val = sunxi_pwm_get_period();
                 ^~~~~~~~~~~~~~~~~~~~
WiringPi/wiringPi/wiringPi.c:1739:9: warning: implicit declaration of function ‘sunxi_pwm_set_act’ [-Wimplicit-function-declaration]
         sunxi_pwm_set_act(value);
         ^~~~~~~~~~~~~~~~~
WiringPi/wiringPi/wiringPi.c: In function ‘wiringPiSetup’:
WiringPi/wiringPi/wiringPi.c:2300:31: error: ‘ORANGEPI’ undeclared (first use in this function); did you mean ‘ERANGE’?
   else if (piGpioLayout () == ORANGEPI)
                               ^~~~~~~~
                               ERANGE
WiringPi/wiringPi/wiringPi.c:2302:20: error: ‘pinToGpioOrangePi’ undeclared (first use in this function); did you mean ‘pinToGpioR2’?
       pinToGpio =  pinToGpioOrangePi;
                    ^~~~~~~~~~~~~~~~~
                    pinToGpioR2
WiringPi/wiringPi/wiringPi.c:2303:16: error: ‘physToGpioOrangePi’ undeclared (first use in this function); did you mean ‘physToGpioR2’?
   physToGpio = physToGpioOrangePi;
                ^~~~~~~~~~~~~~~~~~
                physToGpioR2
WiringPi/wiringPi/wiringPi.c:2304:3: error: ‘physToPin’ undeclared (first use in this function); did you mean ‘physToGpio’?
   physToPin = physToPinOrangePi;
   ^~~~~~~~~
   physToGpio
WiringPi/wiringPi/wiringPi.c:2304:15: error: ‘physToPinOrangePi’ undeclared (first use in this function)
   physToPin = physToPinOrangePi;
               ^~~~~~~~~~~~~~~~~
WiringPi/wiringPi/wiringPi.c:2434:3: error: ‘GPIO_PWM’ undeclared (first use in this function); did you mean ‘GPIO_PADS’?
   GPIO_PWM   = piGpioBase + 0x0020C000 ;
   ^~~~~~~~
   GPIO_PADS
WiringPi/wiringPi/wiringPi.c:2464:3: error: ‘timer’ undeclared (first use in this function); did you mean ‘time’?
   timer = (uint32_t *)mmap(0, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED, fd, GPIO_TIMER) ;
   ^~~~~
   time
WiringPi/wiringPi/wiringPi.c:2474:3: error: ‘timerIrqRaw’ undeclared (first use in this function); did you mean ‘timerclear’?
   timerIrqRaw = timer + TIMER_IRQ_RAW ;
   ^~~~~~~~~~~
   timerclear
WiringPi/wiringPi/wiringPi.c: In function ‘wiringPiSetupSys’:
WiringPi/wiringPi/wiringPi.c:2564:3: error: ‘version’ undeclared (first use in this function); did you mean ‘termio’?
   version = piGpioLayout () ;
   ^~~~~~~
   termio
WiringPi/wiringPi/wiringPi.c:2565:7: error: ‘ORANGEPI’ undeclared (first use in this function); did you mean ‘ERANGE’?
   if (ORANGEPI == version)
       ^~~~~~~~
       ERANGE
WiringPi/wiringPi/wiringPi.c:2567:17: error: ‘pinToGpioOrangePi’ undeclared (first use in this function); did you mean ‘pinToGpioR2’?
    pinToGpio =  pinToGpioOrangePi;
                 ^~~~~~~~~~~~~~~~~
                 pinToGpioR2
WiringPi/wiringPi/wiringPi.c:2568:17: error: ‘physToGpioOrangePi’ undeclared (first use in this function); did you mean ‘physToGpioR2’?
    physToGpio = physToGpioOrangePi;
                 ^~~~~~~~~~~~~~~~~~
                 physToGpioR2
WiringPi/wiringPi/wiringPi.c:2569:4: error: ‘physToPin’ undeclared (first use in this function); did you mean ‘physToGpio’?
    physToPin = physToPinOrangePi;
    ^~~~~~~~~
    physToGpio
WiringPi/wiringPi/wiringPi.c:2569:16: error: ‘physToPinOrangePi’ undeclared (first use in this function)
    physToPin = physToPinOrangePi;
                ^~~~~~~~~~~~~~~~~
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

ranger81, 2021-05-02:
I just came across the same issue and I fixed it like described in #1 

> Added the following line within the _wiringpi = Extension( ... ) block:
define_macros=[('CONFIG_ORANGEPI_PC2',None),('CONFIG_ORANGEPI',None)],
Of course with the corresponding OrangePi Model

> Changed the the line for wiringPiSPISetupMode in binding.i to this:
int wiringPiSPISetupMode (int channel, int port, int speed, int mode) ;

Now it compiled successfully. Hope this helps someone in the future :-)

alexgubanow, 2022-01-17:
@techaura same here, If this is still relevant, you welcome to try official_wiringOP branch. Please close issue, once fix confirmed.