orangepi-xunlong/wiringOP-Python #16: Installation failiure when running  sudo python3 setup.py install

Sheepy3, 2023-12-07:
`sudo python3 setup.py install` returns

```running install
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:146: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
creating wiringpi.egg-info
writing wiringpi.egg-info/PKG-INFO
writing dependency_links to wiringpi.egg-info/dependency_links.txt
writing top-level names to wiringpi.egg-info/top_level.txt
writing manifest file 'wiringpi.egg-info/SOURCES.txt'
file wiringpi.py (for module wiringpi) not found
reading manifest file 'wiringpi.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'wiringpi.py'
warning: no files found matching 'wiringpi_wrap.c'
adding license file 'LICENSE.txt'
writing manifest file 'wiringpi.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-aarch64/egg
running install_lib
running build_py
running build_ext
building '_wiringpi' extension
swigging wiringpi.i to wiringpi_wrap.c
swig -python -threads -o wiringpi_wrap.c wiringpi.i
creating build
creating build/temp.linux-aarch64-cpython-311
creating build/temp.linux-aarch64-cpython-311/wiringOP
creating build/temp.linux-aarch64-cpython-311/wiringOP/devLib
creating build/temp.linux-aarch64-cpython-311/wiringOP/wiringPi
aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -IwiringOP -IwiringOP/wiringPi -IwiringOP/devLib -I/usr/include/python3.11 -c fixUndefFunc.c -o build/temp.linux-aarch64-cpython-311/fixUndefFunc.o "" -DCONFIG_ORANGEPI
aarch64-linux-gnu-gcc: warning: : linker input file unused because linking not done
aarch64-linux-gnu-gcc: error: : linker input file not found: No such file or directory
error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1```

faab-dev, 2024-02-01:
Same problem   Orange Pi Zero 3  (Orange Pi 1.0.2 Jammy with Linux 6.1.31-sun50iw)

ArseniyZh, 2024-02-22:
Exactly the same. Have you found a solution?
@Sheepy3

Sheepy3, 2024-02-22:
nope. still looking for one.

maximilianoteruel, 2024-03-27:
I found the solution... It is necessary fix board name on the setup.py... a quick fix is to add this line:

BOARD = "orangepizero2-h616"

The orange pi zero 3 has the same pinout as the orange pi zero 2

alkopit, 2024-07-30:
Exactly the same.

init-zey, 2024-10-08:
for orangepi zero3, add line `EXTRA_CFLAGS = "-DCONFIG_ORANGEPI_ZERO2"`

Crazy-Child, 2024-11-18:
same problem for zero 2w. how can you solve?