orangepi-xunlong/wiringOP-Python #18: OrangePi Zero 3  wiringpi.wiringPiISR() is Fail

winitboonkliang, 2024-01-30:
OrangePi Zero 3 
2G RAM Model
Linux orangepizero3 5.4.125 #1.0.0 SMP Fri Jun 30 11:59:12 CST 2023 aarch64 GNU/Linux

Step install wiringOP-Python with  python3.9.2

git clone --recursive https://github.com/orangepi-xunlong/wiringOP-Python -b next
cd wiringOP-Python
git submodule update --init --remote

python3 generate-bindings.py > bindings.i
sudo python3 setup.py install

code python3

import wiringpi
PIN_TO_SENSE = 5
def gpio_callback():
    print("GPIO_CALLBACK!")

#wiringpi.wiringPiSetupGpio()  # case 1
wiringpi.wiringPiSetup()  # case 2
wiringpi.pinMode(PIN_TO_SENSE, wiringpi.GPIO.INPUT)
wiringpi.pullUpDnControl(PIN_TO_SENSE, wiringpi.GPIO.PUD_UP)

wiringpi.wiringPiISR(PIN_TO_SENSE, wiringpi.GPIO.INT_EDGE_BOTH, gpio_callback)

while True:
    wiringpi.delay(2000)

Result

case 1 used  wiringpi.wiringPiSetupGpio()  
sudo python3 wiringOP-Python/examples/callback.py
Pin mode failed!

case 2 used wiringpi.wiringPiSetup()
wiringPiISR: unable to open /sys/class/gpio/gpio70/value: No such file or directory

faab-dev, 2024-02-02:
Orange Pi Zero 2
`ls -la /sys/class/gpio/* #before run python script`
Result:
_--w-rw---- 1 root gpio 4096 Feb  2 04:17 /sys/class/gpio/export
lrwxrwxrwx 1 root gpio    0 Feb  2 04:17 /sys/class/gpio/gpiochip0 -> ../../devices/platform/soc/300b000.pinctrl/gpio/gpiochip0
lrwxrwxrwx 1 root gpio    0 Feb  2 04:17 /sys/class/gpio/gpiochip352 -> ../../devices/platform/soc/7022000.pinctrl/gpio/gpiochip352
--w-rw---- 1 root gpio 4096 Feb  2 04:17 /sys/class/gpio/unexport_

`import time
import socketio
import wiringpi
import time
import sys
from wiringpi import GPIO

def gpio_callback():
    print ("GPIO_CALLBACK!")

wiringpi.wiringPiSetupPhys()
wiringpi.wiringPiSetupSys()

PIN_IN = 10
PIN_OUTPUT = 12
wiringpi.pinMode(PIN_OUTPUT, GPIO.OUTPUT)
wiringpi.pinMode(PIN_IN, GPIO.INPUT)
wiringpi.pullUpDnControl(PIN_IN, GPIO.PUD_DOWN)
wiringpi.wiringPiISR(PIN_IN, GPIO.INT_EDGE_BOTH, gpio_callback)`

Result: 
**gpio: Unable to open GPIO edge interface for pin 10: Permission denied**

`ls -la /sys/class/gpio/* #after run python script`
Result:
_--w-rw---- 1 root gpio 4096 Feb  2 05:27 /sys/class/gpio/export
lrwxrwxrwx 1 root root    0 Feb  2 05:27 /sys/class/gpio/gpio10 -> ../../devices/platform/soc/300b000.pinctrl/gpiochip0/gpio/gpio10
lrwxrwxrwx 1 root gpio    0 Feb  2 04:17 /sys/class/gpio/gpiochip0 -> ../../devices/platform/soc/300b000.pinctrl/gpio/gpiochip0
lrwxrwxrwx 1 root gpio    0 Feb  2 04:17 /sys/class/gpio/gpiochip352 -> ../../devices/platform/soc/7022000.pinctrl/gpio/gpiochip352
--w-rw---- 1 root gpio 4096 Feb  2 04:17 /sys/class/gpio/unexport_

Problem: gpio_callback does not working

sanj3k, 2024-04-23:
Hi, it seems we both have the same issue. Did you somehow got it fixed?

haojiawei111, 2024-08-07:
一样的问题，插个眼