orangepi-xunlong/wiringOP-Python #23: import wiringpi 出错，提示No module named '_wiringpi'

czm2333, 2024-04-28:
在zero3上安装，
git clone --recursive https://github.com/orangepi-xunlong/wiringOP-Python -b next
cd wiringOP-Python
git submodule update --init --remote
cd wiringOP-Python
python3 generate-bindings.py > bindings.i
sudo python3 setup.py install
提示安装成功
在import wiringpi 出错，提示No module named '_wiringpi'

closed 2024-06-13 by czm2333

YingzhiLin, 2024-07-27:
我也遇到了同样的问题，您是如何解决的呢？

YingzhiLin, 2024-07-27:
import wiringpi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/src/wiringOP-Python/wiringpi.py", line 15, in <module>
    import _wiringpi
ImportError: /usr/local/lib/python3.10/dist-packages/wiringpi-2.60.1-py3.10-linux-aarch64.egg/_wiringpi.cpython-310-aarch64-linux-gnu.so: undefined symbol: sunxi_pwm_set_clk