orangepi-xunlong/wiringOP-Python #27: WiringOP-Python don't work on Orangepi 5 Plus   ubuntu 22.04 , ubuntu 24.04

DanilSemen, 2024-08-04:
Hello!!
WiringOP-Python don't work on Orangepi 5 Plus   ubuntu 22.04 , ubuntu 24.04
WiringOP-Python has stopped working in the last half a year.
sudo gpioreadall - ok
python3 import wiringpi help  - ok

but

"wiringpi.wiringpisetup "

 given error 
1) if I use the current version WiringOP-Python, given error
 pwm_sunxi 
""
if I use old archives from git,  given error 
"wiringpisetup: mmap (gpio) failed: operation not permitted"

and if i doing

 "echo "BOARD=orangepi5" | sudo tee /etc/orangepi-release"
 "wiringpi.wiringpisetup " - ok , but pin only 26. my  board Orangeoi 5 plus with 40 pin.
 
 if i doing 
 
  "echo "BOARD=orangepi5plus" | sudo tee /etc/orangepi-release" 
  
   "wiringpi.wiringpisetup " -> error "wiringpisetup: mmap (gpio) failed: operation not permitted"
   
   What happening????????

mag1art, 2024-08-07:
can you try reinstall python3.10 and wiringpi, then try run again?

DanilSemen, 2024-08-09:
> can you try reinstall python3.10 and wiringpi, then try run again?

Its dont work